f0cal/google-coral #84: Pimping boot, background, loading

laurentdebricon, 2020-08-09:
Hello,

i am trying to pimp the google coral with my logo // color scheme, to sell it to a client. So far a script is launching at startup (thanks to https://github.com/google-coral/edgetpu/issues/120) 

I am looking for solutions :
- how to put "quiet" in the boot loading (to hide the linux kernel message at boot (i tried the debian way but no /etc/default/grub ))
- even better add my splash logo (or with plymouth)
- change the default blue background with my logo (it takes 2 secondes of the blue background before the startup detection launch full screen)

Best regards

Namburger, 2020-08-10:
@laurentdebricon Hi :)
In general, embeded linux such as the dev board doesn't have grub, we use something call u-boot. You can follow this answer to modify kernel command line:

Download boot.txt:
```
$ curl https://coral.googlesource.com/build/+/refs/heads/docker/boot.txt\?format\=TEXT | base64 --decode | tee boot.txt > /dev/null
```
Install mkimage:
```
$ sudo apt install u-boot-tools
```
Make your necessary changes in the cmdline="" line, for this example, we need to add "quiet loglevel=0":
```
cmdline=<preexsisting> + quiet loglevel=0
```
compile to boot.scr:
```
$ mkimage -A arm -T script -O linux -d boot.txt boot.scr
```
replace old boot image file
```
$ mv boot.scr > /boot
```
Reboot and the new kernel params should be loaded.

laurentdebricon, 2020-08-11:
Thank you @Namburger , 

- Four proud Tux are still displayed, but quiet, it's a good start. In this tutorial https://developer.toradex.com/knowledge-base/u-boot-splash-screen#Practical_Example
you can load a picture at runtime, don't know if the uboot of the Coral supports that ... 

- I have found the location of the desktop background :
`/usr/share/weston/background.png`

closed 2020-08-18 by laurentdebricon

Namburger, 2020-08-19:
@laurentdebricon ohh, you can change the background-image config in `/etc/xdg/weston/weston.ini`:
```
mendel@lime-rabbit:~/sdcard/testing$ cat /etc/xdg/weston/weston.ini 
[core]
gbm-format=argb8888
shell=desktop-shell.so
modules=systemd-notify.so
xwayland=true
idle-time=0

[shell]
background-image=/usr/share/weston/background.png
background-type=scale
allow-zap=false

[keyboard]
vt-switching=true

[output]
name=HDMI-A-1
mode=1920x1080
```