f0cal/google-coral #85: Fan Noise reduction solutions

laurentdebricon, 2020-08-18:
How to decrease the fan noise ?
i read about fan starting point but I am looking for fan noise reduction solutions ! Fan replacement tutorial. Any idea ?

This case makes the sound worse. 
https://kksb-cases.com/products/kksb-coral-dev-board-case-aluminum

Namburger, 2020-08-18:
Humn, I'm not actually sure about fan noise... but the J6 connector has a power/ground pin if you want to replace with another fan. I'm not sure if replacing the fan would make a difference in noise 
fyi I also have one of those case :D and also, this isn't our official channel, suggest future issues to be sent to [this repo](https://github.com/google-coral/edgetpu/issues) for library related questions and [google-coral stackoverflow channel](https://stackoverflow.com/questions/63428818/building-tensorflow-lite-for-edge-tpu-fails) for questions like these!

Namburger, 2020-08-19:
@laurentdebricon 
Here's a mini fanless solution that should be coming really really soon :) 
https://coral.ai/products/dev-board-mini

If you're just doing mask detection and not much CPU usage, I think the mini would be a great solution. I don't have exact date/price for it, unfortunately :/