f0cal/google-coral #86: Google Coral Dev Board not recognized by host computer

wjsgus95, 2020-09-08:
* Doc you were trying to follow: https://coral.ai/docs/dev-board/get-started/#12-execute-the-flash-script
* Your host OS: Linux Mint 20 64-bit
* Your Python3 version: Python 3.8.2

# What I ran
`dmesg | grep ttyUSB`

# What the docs said should happen
`dmesg | grep ttyUSB`
is the command that's on the doc and I'm getting nothing when I'm supposed to get
```
[ 6437.706335] usb 2-13.1: cp210x converter now attached to ttyUSB0
[ 6437.708049] usb 2-13.1: cp210x converter now attached to ttyUSB1
```

I have no idea what the problem is.
- I have added my username to _plugdev_ and _dialout_.
- There's orange and green light on on my Dev Board.
- My fastboot version is 30. something.

# What actually happened

wjsgus95, 2020-09-08:
I tried to reflash with the recovery image in sd card but I can't do that because I can't connect through the serial console port. Any help would be much appreciated.

Namburger, 2020-09-09:
Hi!
if you plug the hdmi cable in the board, does it show any outputs when you turn the board on? 

FYI, this isn't our official repo, for future issues, please open it here: https://github.com/google-coral/edgetpu/issues

wjsgus95, 2020-09-09:
Okay. I'll post future issues in the link you put.

Thanks for the suggestion. 

I plugged in the hdmi cable and it gave me an empty desktop background and I was able to open the terminal.

wjsgus95, 2020-09-09:
So very occasionally "$ mdt shell" works, so I managed to open a ssh connection via wifi. But still,

`dmesg | grep ttyUSB`

`fastboot --devices`

`mdt devices`

gives me nothing.

p.s. My computer's using cdc_acm module instead of cp210x as the usb driver, and I can access the dev board through ttyACM0

Namburger, 2020-09-09:
@wjsgus95 I see, `cdc_acm` module is actually standard and is what we use over usb-c, cp210x is only for serial so it's not completely necessary.
Although with `cdc_acm` we noticed that sometimes this becomes an issue and the fix.... is to actually just unplug all usb devices on your host machine and reboot :P 

FYI, if you can still mdt into the board, I suggest setting up ssh via wifi channel instead of usb, mdt is just a friendly wrapper around ssh

wjsgus95, 2020-09-14:
Thank you Nam. I'll close this thread for now. I've managed to set up ssh connection so I'll be using that for now. 

I'll post another issue on the right repo next time. Cheers

closed 2020-09-14 by wjsgus95