f0cal/google-coral #87: Do the changes in TPU's frequency (STD, MAX) work in the devboard? what happens to the CPU's frequency during these changes?

aizzaac, 2020-09-21:
* Doc you were trying to follow: 
[image]

* Your host OS: WINDOWS10

* Your Python3 version: PYTHON3

# What I ran
I changed the frequency from STD to MAX. Then I ran inference. I do not see difference when changing frequencies.

# What the docs said should happen
I do not fully understand the docs. I am not sure whether the change in frequency works in the devboard or not.

# What actually happened
There is not difference in inference speed when changing frequencies.

Namburger, 2020-09-22:
The std library should only throttles if the tpu reach a certain temperature threshold, under normal conditions, the max should be the same as std.
https://github.com/google-coral/libedgetpu/blob/642a289e4fb3932422b577a1c4c2bfc792ed1d15/tflite/edgetpu_manager_direct.cc#L124

cheers

aizzaac, 2020-09-24:
In what category does the devboard fit: USB or PCIE?

Namburger, 2020-09-24:
@aizzaac pcie for the dev board, usb is this one https://coral.ai/products/accelerator/