f0cal/google-coral #89: Trying to install Coral Enviro Board to R.Pi3B

neilh10, 2020-09-28:
Followed the instructions, and the display reads out "nam" for actual readings.

Purchased the Coral Enviro Board from Mouser. 
I'm wondering, shouldn't there be an overaly in /boot/overlays for the CoralEnviroBoard.

Namburger, 2020-09-28:
@neilh10 I'm sorry, what's the actual issue?
There should be a /boot/overlay.txt in mendel

neilh10, 2020-09-29:
Hello @Namburger , thankyou for getting back. 
I followed https://coral.ai/docs/enviro-board/get-started/  and when it ran the display said "nam" for actual readings. (not nan)
While I'm software engineer in uP Iot, and I have done some BeagleBOne - this is a little school project with the R.Pi and a neighbour.  
So it just seems to me this is a rabbit hole - there aren't any links describing what to do, or look out for - it seems its supposed to just work, and it isn't.  I checked out possible other hints of how to solve it #21 , #38

If the board isn't supported anymore I should just use one of the others that are out there PIM486

Trying to guess at some debuggin ..

pi@rpi3nhab:~ $ ls /boot/overlay.txt
ls: cannot access '/boot/overlay.txt': No such file or directory

pi@rpi3nhab:~ $ ls /boot
bcm2708-rpi-b.dtb         bcm2710-rpi-cm3.dtb  fixup4x.dat   LICENCE.broadcom
bcm2708-rpi-b-plus.dtb    bcm2711-rpi-4-b.dtb  fixup_cd.dat  overlays
bcm2708-rpi-b-rev1.dtb    bcm2711-rpi-cm4.dtb  fixup.dat     start4cd.elf
bcm2708-rpi-cm.dtb        bootcode.bin         fixup_db.dat  start4db.elf
bcm2708-rpi-zero.dtb      cmdline.txt          fixup_x.dat   start4.elf
bcm2708-rpi-zero-w.dtb    config.txt           issue.txt     start4x.elf
bcm2709-rpi-2-b.dtb       COPYING.linux        kernel7.img   start_cd.elf
bcm2710-rpi-2-b.dtb       fixup4cd.dat         kernel7l.img  start_db.elf
bcm2710-rpi-3-b.dtb       fixup4.dat           kernel8.img   start.elf
bcm2710-rpi-3-b-plus.dtb  fixup4db.dat         kernel.img    start_x.elf

I thought overlays are .dtbo ?  ~ but can't see anything that relates to this board. Is there a way of listing the boards EEPROM to be able to verify what its requiring 

rpi3nhab:~ $ ls /boot/overlays/
act-led.dtbo                            mcp2515-can1.dtbo
adau1977-adc.dtbo                       mcp3008.dtbo
adau7002-simple.dtbo                    mcp3202.dtbo
ads1015.dtbo                            mcp342x.dtbo
ads1115.dtbo                            media-center.dtbo
ads7846.dtbo                            merus-amp.dtbo
adv7282m.dtbo                           midi-uart0.dtbo
adv728x-m.dtbo                          midi-uart1.dtbo
akkordion-iqdacplus.dtbo                miniuart-bt.dtbo
allo-boss-dac-pcm512x-audio.dtbo        mmc.dtbo
allo-digione.dtbo                       mpu6050.dtbo
allo-katana-dac-audio.dtbo              mz61581.dtbo
allo-piano-dac-pcm512x-audio.dtbo       ov5647.dtbo
allo-piano-dac-plus-pcm512x-audio.dtbo  ov7251.dtbo
anyspi.dtbo                             ov9281.dtbo
apds9960.dtbo                           overlay_map.dtb
applepi-dac.dtbo                        papirus.dtbo
at86rf233.dtbo                          pibell.dtbo
audioinjector-addons.dtbo               piglow.dtbo
audioinjector-isolated-soundcard.dtbo   piscreen2r.dtbo
audioinjector-ultra.dtbo                piscreen.dtbo
audioinjector-wm8731-audio.dtbo         pisound.dtbo
audiosense-pi.dtbo                      pitft22.dtbo
audremap.dtbo                           pitft28-capacitive.dtbo
balena-fin.dtbo                         pitft28-resistive.dtbo
cma.dtbo                                pitft35-resistive.dtbo
dht11.dtbo                              pps-gpio.dtbo
dionaudio-loco.dtbo                     pwm-2chan.dtbo
dionaudio-loco-v2.dtbo                  pwm.dtbo
disable-bt.dtbo                         pwm-ir-tx.dtbo
disable-wifi.dtbo                       qca7000.dtbo
dpi18.dtbo                              README
dpi24.dtbo                              rotary-encoder.dtbo
draws.dtbo                              rpi-backlight.dtbo
dwc2.dtbo                               rpi-cirrus-wm5102.dtbo
dwc-otg.dtbo                            rpi-dac.dtbo
enc28j60.dtbo                           rpi-display.dtbo
enc28j60-spi2.dtbo                      rpi-ft5406.dtbo
exc3000.dtbo                            rpi-poe.dtbo
fe-pi-audio.dtbo                        rpi-proto.dtbo
goodix.dtbo                             rpi-sense.dtbo
googlevoicehat-soundcard.dtbo           rpi-tv.dtbo
gpio-fan.dtbo                           rpivid-v4l2.dtbo
[... 47 lines cut ...]
justboom-both.dtbo                      upstream-pi4.dtbo
justboom-dac.dtbo                       vc4-fkms-v3d.dtbo
justboom-digi.dtbo                      vc4-kms-kippah-7inch.dtbo
ltc294x.dtbo                            vc4-kms-v3d.dtbo
max98357a.dtbo                          vc4-kms-v3d-pi4.dtbo
maxtherm.dtbo                           vga666.dtbo
mbed-dac.dtbo                           w1-gpio.dtbo
mcp23017.dtbo                           w1-gpio-pullup.dtbo
mcp23s17.dtbo                           w5500.dtbo
mcp2515-can0.dtbo                       wittypi.dtbo

pi@rpi3nhab:~ $ cat /proc/version
Linux version 5.4.51-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1333 SMP Mon Aug 10 16:45:19 BST 2020

pi@rpi3nhab:~ $  i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: 30 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- 45 -- -- -- 49 -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- 76 --
pi@rpi3nhab:~ $

neilh10, 2020-09-29:
Hello gosh my bad eyes - it is actually showing "nan" and for some reason with the small type I was seeing "nam" - I've treated to correct all the above references.

Namburger, 2020-09-29:
@neilh10 ahhh, my apologies, I read the title of your issue wrong and completely thought that you're trying to do something else which leads me to all type of confusion from the info that you gave :P 
Scatch what I said regarding /boot/verlays.txt, that exists on mendel OS which is the official OS on the dev board!

>  it seems its supposed to just work, and it isn't

It is supposed to just works so this could be an issue :) Here are a couple of suggestions:

1) Make sure SPI and I2C are enable. This can be done with `sudo raspi-config` and go to *Interfacing Options*, then enable SPI + I2C and reboot.

2) Could you also apply this fix: https://github.com/f0cal/google-coral/issues/21#issuecomment-691426061?
 
```
sudo vim /usr/lib/python3/dist-packages/coral/enviro/board.py
```
go to [this line](https://coral.googlesource.com/coral-cloud/+/refs/heads/master/python/coral-enviro/coral/enviro/board.py#73) and change:
```
self._display = ssd1306(serial_interface=spi(cs_high=True),
```
to this:
```
self._display = ssd1306(serial_interface=spi(),
```
Then `:wq!` to save the file and rerun the demo

neilh10, 2020-09-29:
Hello
Yes I had already gone through with ~ udo raspi-config and go to Interfacing Options, then enable SPI + I2C 
and no effect. Just repeated to make sure.

The ssd1306  looks like its already been incorporated - file attached, and this is what I actually see

            # Default to RPi.GPIO in luma and defaults GPIO.
            self._display = ssd1306(serial_interface=spi(),
                                    gpio=noop(), height=32, rotate=2)

[board.py.txt](https://github.com/f0cal/google-coral/files/5300425/board.py.txt)

rebooted and tried again. Still same problem

Namburger, 2020-09-29:
whoopies, you have to add that cs_high=True to the serial_interface argument:
```
# Default to RPi.GPIO in luma and defaults GPIO.
        self._display = ssd1306(serial_interface=spi(csi_high=True),
                                gpio=noop(), height=32, rotate=2)
```
Sorry, I reversed the order on that previous comment...)

neilh10, 2020-10-05:
Oh thanks for the pointer. In using it as a tutorial this morning, still doesn't work
as specified has error, can't find option csi_high
From https://luma-core.readthedocs.io/en/latest/interface.html
seems like for spi() there is option cs_high
# Default to RPi.GPIO in luma and defaults GPIO.
        self._display = ssd1306(serial_interface=spi(cs_high=True),
                                gpio=noop(), height=32, rotate=2)

This now runs, However this still gives same "nan"

For such a simple program (with lots happening in the libs), seems like something is missing. Either specifying what version of raspberryPi OS it was tested on, OR .... ....

neilh10, 2020-10-06:
I just rebuilt a raspberry from new on a uSD, enabled the SPI and I2C with "sudo raspi-config", added the cs_high=True. 
Still same problem 'nan'

Luisibear98, 2020-11-17:
Im having the same trouble here! 
My OS version is:

- Linux version 5.4.72-v7+ (dom@buildbot) (gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)) #1356 SMP Thu Oct 22 13:56:54 BST 2020

cheers!

Luisibear98, 2020-11-17:
I think problems comes when board.py tries to look for the sensors.

The path /sys/bus/iio/devices/ seems to be empty

neilh10, 2020-11-17:
I've gone with https://shop.pimoroni.com/products/enviro?variant=31155658489939 PIM486 and it worked straight away.  
Be interested if you figure it out.

Luisibear98, 2020-11-18:
> I've gone with https://shop.pimoroni.com/products/enviro?variant=31155658489939 PIM486 and it worked straight away.
> Be interested if you figure it out.
Thanks for the idea!

Finally decided to go for a custom version of the code. It is a modification from the original project adding more libraries to control the sensors. Now I'm getting all the values! let me know if any of you needs the code.

jeffreycunn, 2021-02-07:
> Finally decided to go for a custom version of the code. It is a modification from the original project adding more libraries to control the sensors. Now I'm getting all the values! let me know if [...]

What did you end up doing? Which libraries?

Luisibear98, 2021-02-08:
> What did you end up doing? Which libraries?

Hi! sorry for the delay, I made a custom repo for that solution. Check my profile! :D

Luisibear98, 2021-02-08:
> What did you end up doing? Which libraries?

https://github.com/Luisibear98/google-coral.git Thats the repo! just enter on the folder and install requirements!

jeffreycunn, 2021-02-15:
Even when I download your repo, the sensor will not display RH or light level....this is really perplexing to me.  I think there is something wrong at a lower level.

Luisibear98, 2021-02-15:
> Even when I download your repo, the sensor will not display RH or light level....this is really perplexing to me. I think there is something wrong at a lower level.

Did you cloned my repo and Installed the requirements.txt? Im using python3 and it is working :).
I have updated the enviro_board class. What fo yoy get when you execute it?
This is what I get:
{'CPU': 46.698, 'temperature': 35.163421630859375, 'humidity': 35.8489990234375, 'ambient_light': 53.86, 'pressure': 945.9359053930328}

Luisibear98, 2021-02-15:
> Even when I download your repo, the sensor will not display RH or light level....this is really perplexing to me. I think there is something wrong at a lower level.

In addition, could you test the test.py script I did?
I have added it now. 
Just do "python3 test.py -d 0x45" 
Does it print the  lux?
If so just try again the enviro_board.py