f0cal/google-coral #92: Eagle enterprise to Day 13 reflashing failure?

startupstarter, 2020-12-07:
* Doc you were trying to follow: <t>
* Your host OS: <Mac catalina>
* Your Python3 version: <Python3>

Bash flash.sh results in: 

downloading of 134217728 bytes finished
can't find partition: fbmisc, dump the partition table
idx 0, ptn 0 name='gpt' start=0 len=2048
idx 1, ptn 0 name='tos' start=0 len=2047
idx 2, ptn 0 name='bootloader0' start=66 len=4094
idx 3, ptn 1 name='boot' start=16384 len=262144
idx 4, ptn 2 name='misc' start=278528 len=2048
idx 5, ptn 3 name='rootfs' start=294912 len=14958592
cannot get the partion info for fbmisc
fastboot_get_lock_stat: error in get mmc part
writing to partition 'boot'
Initializing 'boot'
switch to partitions #0, OK
mmc0(part 0) is current device
Writing 'boot'

MMC write: dev # 0, block # 16384, count 262144 ... 262144 blocks written: OK
Writing 'boot' DONE!
request 000000007bf34880 was not queued to ep1in-bulk
request 000000007bf34880 was not queued to ep1in-bulk

Any idea how to downgrade to mendel 4.0?

Namburger, 2020-12-08:
What board do you have?

mike122112, 2020-12-09:
> What board do you have?

Dev board 1G with ">9410" serial number