f0cal/google-coral #95: Errors on install of python3-coral-enviro with Raspberry Pi Zero W and NaN for Temperature and and Light Level

jeffreycunn, 2021-02-06:
* Doc you were trying to follow: https://coral.ai/docs/enviro-board/get-started/#2-install-the-python-library
* Your host OS: Raspberry Pi OS 32 bit on a Raspberry Pi Zero W (latest stable image) and Coral Environmental Sensor Board
* Your Python3 version: Python 3.7.3

# What I ran
sudo apt install python3-coral-enviro

# What the docs said should happen

# What actually happened
Reading package lists... Done
Building dependency tree       
Reading state information... Done

The following additional packages will be installed:
  coral-enviro-drivers-dkms dkms python3-coral-cloudiot python3-cryptoauthlib
  python3-luma-core python3-luma-oled python3-paho-mqtt python3-periphery
  python3-smbus2 raspberrypi-kernel-headers
Suggested packages:
  python3-apport menu
The following NEW packages will be installed:
  coral-enviro-drivers-dkms dkms python3-coral-cloudiot python3-coral-enviro
  python3-cryptoauthlib python3-luma-core python3-luma-oled python3-paho-mqtt
  python3-periphery python3-smbus2 raspberrypi-kernel-headers
0 upgraded, 11 newly installed, 0 to remove and 0 not upgraded.
Need to get 27.9 MB of archives.
After this operation, 181 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel-headers armhf 1.20210201-1 [27.6 MB]
Get:3 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf coral-enviro-drivers-dkms all 1.5 [17.0 kB]
Get:5 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-cryptoauthlib armhf 20190304-3 [62.3 kB]
Get:6 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-coral-cloudiot all 1.2 [6,816 B]
Get:2 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf dkms all 2.6.1-4 [74.4 kB]
Get:7 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-smbus2 all 0.2.2-1 [8,456 B]
Get:8 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-luma-core all 1.8.3-1 [34.8 kB]
Get:9 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-luma-oled all 3.2.0-1 [13.8 kB]
Get:10 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-periphery all 1.1.1-1 [13.9 kB]
Get:11 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-coral-enviro all 1.1 [5,468 B]
Get:4 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf python3-paho-mqtt all 1.4.0-1 [44.9 kB]
Fetched 27.9 MB in 2min 15s (207 kB/s)                                         
Selecting previously unselected package dkms.
(Reading database ... 98526 files and directories currently installed.)
Preparing to unpack .../00-dkms_2.6.1-4_all.deb ...
Unpacking dkms (2.6.1-4) ...
Selecting previously unselected package raspberrypi-kernel-headers.
Preparing to unpack .../01-raspberrypi-kernel-headers_1.20210201-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20210201-1) ...
Selecting previously unselected package coral-enviro-drivers-dkms.
Preparing to unpack .../02-coral-enviro-drivers-dkms_1.5_all.deb ...
Unpacking coral-enviro-drivers-dkms (1.5) ...
Selecting previously unselected package python3-paho-mqtt.
Preparing to unpack .../03-python3-paho-mqtt_1.4.0-1_all.deb ...
Unpacking python3-paho-mqtt (1.4.0-1) ...
Selecting previously unselected package python3-cryptoauthlib.
Preparing to unpack .../04-python3-cryptoauthlib_20190304-3_armhf.deb ...
Unpacking python3-cryptoauthlib (20190304-3) ...
Selecting previously unselected package python3-coral-cloudiot.
Preparing to unpack .../05-python3-coral-cloudiot_1.2_all.deb ...
Unpacking python3-coral-cloudiot (1.2) ...
Selecting previously unselected package python3-smbus2.
Preparing to unpack .../06-python3-smbus2_0.2.2-1_all.deb ...
Unpacking python3-smbus2 (0.2.2-1) ...
Selecting previously unselected package python3-luma-core.
Preparing to unpack .../07-python3-luma-core_1.8.3-1_all.deb ...
Unpacking python3-luma-core (1.8.3-1) ...
Selecting previously unselected package python3-luma-oled.
Preparing to unpack .../08-python3-luma-oled_3.2.0-1_all.deb ...
Unpacking python3-luma-oled (3.2.0-1) ...
Selecting previously unselected package python3-periphery.
Preparing to unpack .../09-python3-periphery_1.1.1-1_all.deb ...
Unpacking python3-periphery (1.1.1-1) ...
Selecting previously unselected package python3-coral-enviro.
Preparing to unpack .../10-python3-coral-enviro_1.1_all.deb ...
Unpacking python3-coral-enviro (1.1) ...
Setting up dkms (2.6.1-4) ...
Setting up raspberrypi-kernel-headers (1.20210201-1) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.11+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.11-v7+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.11-v7l+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.11-v8+
Setting up python3-paho-mqtt (1.4.0-1) ...
Setting up python3-periphery (1.1.1-1) ...
Setting up python3-smbus2 (0.2.2-1) ...
Setting up python3-cryptoauthlib (20190304-3) ...
Setting up coral-enviro-drivers-dkms (1.5) ...
Loading new coral-enviro-drivers-1.5 DKMS files...
It is likely that 5.10.11+ belongs to a chroot's host
Building for 5.10.11+, 5.10.11-v7+, 5.10.11-v7l+ and 5.10.11-v8+
Building initial module for 5.10.11+
Error! Bad return status for module build on kernel: 5.10.11+ (armv6l)
Consult /var/lib/dkms/coral-enviro-drivers/1.5/build/make.log for more information.
dpkg: error processing package coral-enviro-drivers-dkms (--configure):
 installed coral-enviro-drivers-dkms package post-installation script subprocess returned error exit status 10
Setting up python3-coral-cloudiot (1.2) ...
Setting up python3-luma-core (1.8.3-1) ...
Setting up python3-luma-oled (3.2.0-1) ...
dpkg: dependency problems prevent configuration of python3-coral-enviro:
 python3-coral-enviro depends on coral-enviro-drivers-dkms; however:
  Package coral-enviro-drivers-dkms is not configured yet.

dpkg: error processing package python3-coral-enviro (--configure):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 coral-enviro-drivers-dkms
 python3-coral-enviro
E: Sub-process /usr/bin/dpkg returned an error code (1)

# After installing I ran:  
python3 enviro_demo.py 
Temperature and Pressure displayed on the OLED, but light levels and RH were registered as NaN

javiercanadillas, 2021-03-13:
Hi,

This is happening to me as well, here's what I did:

### What I have
- Following official [Get Started](https://coral.ai/docs/enviro-board/get-started/)
- Raspberry Pi OS 32 bit on a Raspberry Pi Zero W (latest stable image) and Coral Environmental Sensor Board

### What I did
My process of setting up the Pi has been as follows:
- Etched a fresh SD Card with the latest Raspberry Pi OS image, using Raspberry Pi Imager
- Inserted the card in the Pi
- Did the following (as per [this issue](https://github.com/f0cal/google-coral/issues/21#issuecomment-506439134)):
``` bash
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
sudo reboot now
sudo echo "deb https://packages.cloud.google.com/apt coral-cloud-stable main" | sudo tee /etc/apt/sources.list.d/coral-cloud.list
sudo curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -
sudo apt update
```

### What has happened
After updating sources, finally run:
```bash
sudo apt install python3-coral-enviro
```
```text
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  coral-enviro-drivers-dkms dkms python3-coral-cloudiot python3-cryptoauthlib python3-luma-core python3-luma-oled python3-paho-mqtt python3-periphery
  python3-smbus2 raspberrypi-kernel-headers
Suggested packages:
  python3-apport menu
The following NEW packages will be installed:
  coral-enviro-drivers-dkms dkms python3-coral-cloudiot python3-coral-enviro python3-cryptoauthlib python3-luma-core python3-luma-oled python3-paho-mqtt
  python3-periphery python3-smbus2 raspberrypi-kernel-headers
0 upgraded, 11 newly installed, 0 to remove and 0 not upgraded.
Need to get 27.9 MB of archives.
After this operation, 181 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel-headers armhf 1.20210303-1 [27.6 MB]
Get:2 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf dkms all 2.6.1-4 [74.4 kB]                                                       
Get:3 http://ftp.cica.es/mirrors/Linux/raspbian/raspbian buster/main armhf python3-paho-mqtt all 1.4.0-1 [44.9 kB]
Get:4 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf coral-enviro-drivers-dkms all 1.5 [17.0 kB]
Get:5 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-cryptoauthlib armhf 20190304-3 [62.3 kB]                  
Get:6 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-coral-cloudiot all 1.2 [6,816 B]
Get:7 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-smbus2 all 0.2.2-1 [8,456 B]
Get:8 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-luma-core all 1.8.3-1 [34.8 kB]
Get:9 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-luma-oled all 3.2.0-1 [13.8 kB]
Get:10 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-periphery all 1.1.1-1 [13.9 kB]
Get:11 https://packages.cloud.google.com/apt coral-cloud-stable/main armhf python3-coral-enviro all 1.1 [5,468 B]
Fetched 27.9 MB in 13s (2,224 kB/s)                                                                                                                         
Selecting previously unselected package dkms.
(Reading database ... 98611 files and directories currently installed.)
Preparing to unpack .../00-dkms_2.6.1-4_all.deb ...
Unpacking dkms (2.6.1-4) ...
Selecting previously unselected package raspberrypi-kernel-headers.
Preparing to unpack .../01-raspberrypi-kernel-headers_1.20210303-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20210303-1) ...
Selecting previously unselected package coral-enviro-drivers-dkms.
Preparing to unpack .../02-coral-enviro-drivers-dkms_1.5_all.deb ...
Unpacking coral-enviro-drivers-dkms (1.5) ...
Selecting previously unselected package python3-paho-mqtt.
Preparing to unpack .../03-python3-paho-mqtt_1.4.0-1_all.deb ...
Unpacking python3-paho-mqtt (1.4.0-1) ...
Selecting previously unselected package python3-cryptoauthlib.
Preparing to unpack .../04-python3-cryptoauthlib_20190304-3_armhf.deb ...
Unpacking python3-cryptoauthlib (20190304-3) ...
[... 27 lines cut ...]
Setting up python3-cryptoauthlib (20190304-3) ...
Setting up coral-enviro-drivers-dkms (1.5) ...
Loading new coral-enviro-drivers-1.5 DKMS files...
It is likely that 5.10.17+ belongs to a chroot's host
Building for 5.10.17+, 5.10.17-v7+, 5.10.17-v7l+ and 5.10.17-v8+
Building initial module for 5.10.17+
Error! Bad return status for module build on kernel: 5.10.17+ (armv6l)
Consult /var/lib/dkms/coral-enviro-drivers/1.5/build/make.log for more information.
dpkg: error processing package coral-enviro-drivers-dkms (--configure):
 installed coral-enviro-drivers-dkms package post-installation script subprocess returned error exit status 10
Setting up python3-coral-cloudiot (1.2) ...
Setting up python3-luma-core (1.8.3-1) ...
Setting up python3-luma-oled (3.2.0-1) ...
dpkg: dependency problems prevent configuration of python3-coral-enviro:
 python3-coral-enviro depends on coral-enviro-drivers-dkms; however:
  Package coral-enviro-drivers-dkms is not configured yet.

dpkg: error processing package python3-coral-enviro (--configure):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 coral-enviro-drivers-dkms
 python3-coral-enviro
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

lucaired, 2021-03-20:
Hello,
I also have the same problem. 

## What I have 

-  I'm using https://coral.ai/docs/enviro-board/get-started/
- On a Raspberry Pi 3 Model B v1.2 with 
Linux raspberrypi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux

## What I did

```
echo "deb https://packages.cloud.google.com/apt coral-cloud-stable main" | sudo tee /etc/apt/sources.list.d/coral-cloud.list

curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -

sudo apt update

sudo apt upgrade

sudo apt install python3-coral-enviro
```

## What has happened

```
pi@raspberrypi:~ $ sudo apt install python3-coral-enviro
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-coral-enviro is already the newest version (1.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Setting up coral-enviro-drivers-dkms (1.5) ...
Removing old coral-enviro-drivers-1.5 DKMS files...

------------------------------
Deleting module version: 1.5
completely from the DKMS tree.
------------------------------
Done.
Loading new coral-enviro-drivers-1.5 DKMS files...
It is likely that 5.10.17-v7+ belongs to a chroot's host
Building for 5.10.17+, 5.10.17-v7+, 5.10.17-v7l+ and 5.10.17-v8+
Building initial module for 5.10.17+
Error! Bad return status for module build on kernel: 5.10.17+ (armv7l)
Consult /var/lib/dkms/coral-enviro-drivers/1.5/build/make.log for more information.
dpkg: error processing package coral-enviro-drivers-dkms (--configure):
 installed coral-enviro-drivers-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of python3-coral-enviro:
 python3-coral-enviro depends on coral-enviro-drivers-dkms; however:
  Package coral-enviro-drivers-dkms is not configured yet.

dpkg: error processing package python3-coral-enviro (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 coral-enviro-drivers-dkms
 python3-coral-enviro
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

rjpadilla, 2021-05-24:
Hello, I am having the same issue when installing `python3-coral-enviro`.

## What I Have
Building from a new raspberry pi image: https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2021-03-25/2021-03-04-raspios-buster-armhf.zip.torrent

## What I Did
Grabbed the debian package using `deb https://packages.cloud.google.com/apt coral-cloud-stable main`
Installed `python3-coral-enviro`

## What Has Happened
```It is likely that 5.10.17-v7l+ belongs to a chroot's host
Building for 5.10.17+, 5.10.17-v7+, 5.10.17-v7l+ and 5.10.17-v8+
Building initial module for 5.10.17+
Error! Bad return status for module build on kernel: 5.10.17+ (armv7l)
Consult /var/lib/dkms/coral-enviro-drivers/1.5/build/make.log for more information.
dpkg: error processing package coral-enviro-drivers-dkms (--configure):
 installed coral-enviro-drivers-dkms package post-installation script subprocess returned error exit status 10
Setting up python3-coral-cloudiot (1.2) ...
Setting up python3-luma-core (1.8.3-1) ...
Setting up python3-luma-oled (3.2.0-1) ...
dpkg: dependency problems prevent configuration of python3-coral-enviro:
 python3-coral-enviro depends on coral-enviro-drivers-dkms; however:
  Package coral-enviro-drivers-dkms is not configured yet.

dpkg: error processing package python3-coral-enviro (--configure):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 coral-enviro-drivers-dkms
 python3-coral-enviro
E: Sub-process /usr/bin/dpkg returned an error code (1)

msiemens, 2021-09-17:
Same here. But I found someone asking the same question on StackOverflow: https://stackoverflow.com/questions/66084814/what-is-causing-how-can-i-fix-build-error-with-sudo-apt-install-python3-coral-e. The top answer mentions that the compilation fails because the referenced Linux IIO driver API has changed but Google hasn't updated their driver. I modified `/usr/src/coral-enviro-drivers-1.5/adc/ti-ads1015.c` by commenting out the two offending lines (797-798 for me):

```c
        // .postenable  = iio_triggered_buffer_postenable,
        // .predisable  = iio_triggered_buffer_predisable,
```

Running `sudo apt install python3-coral-enviro` a second time worked and produced a working driver 🙂

A real fix would involve Google updating their device drivers, but I don't really see this happening considering the last commit is almost a year old (see https://coral.googlesource.com/coral-cloud/)…

MarkusBansky, 2022-04-22:
Hi everyone, I wanted to add my workaround in addition to the comment @msiemens made. As of 2022, the environment board example did not work on RP as there were some problems with the SPI interface because Google did not update the library.

After successfully installing the `python3-coral-enviro` go and upgrade the `luma` dependency for `oled`:

```bash
$ sudo apt-get update
$ sudo apt-get install python3 python3-pip python3-pil libjpeg-dev zlib1g-dev libfreetype6-dev liblcms2-dev libopenjp2-7 libtiff5 -y
$ sudo -H pip3 install --upgrade luma.oled
```

Then reboot the board and you can run the example in `cd /usr/lib/python3/dist-packages/coral/enviro
` without a problem.

ITotallyKnowNothing, 2022-06-02:
Did what msiemens suggested, edited the ti-ads1015.c source file in a sudo su terminal running Geany, commented out line 797 & 798 /*...*/
Saved and then re-run the apt install python3-coral-enviro and it worked (still in a sudo su). You see a lot more happening as well during the install.
The 2 definitions

.postenable  = iio_triggered_buffer_postenable,
.predisable  = iio_triggered_buffer_predisable,

are not even used let alone defined, comment them out and it all worked.
This was on a RPi400.

morbidnomad, 2023-01-17:
> Hi everyone, I wanted to add my workaround in addition to the comment @msiemens made. As of 2022, the environment board example did not work on RP as there were some problems with the SPI interface because Google did not update the library.
> 
> After successfully installing the `python3-coral-enviro` go and upgrade the `luma` dependency for `oled`:
> 
> ```shell
> $ sudo apt-get update
> $ sudo apt-get install python3 python3-pip python3-pil libjpeg-dev zlib1g-dev libfreetype6-dev liblcms2-dev libopenjp2-7 libtiff5 -y
> $ sudo -H pip3 install --upgrade luma.oled
> ```
> 
> Then reboot the board and you can run the example in `cd /usr/lib/python3/dist-packages/coral/enviro ` without a problem.

That absolutely worked for me, looks like an issue with the oled driver. Thanks

miroslavandrushchak, 2024-05-21:
> Hello, I also have the same problem.
>
> ## What I have
> * I'm using https://coral.ai/docs/enviro-board/get-started/
> * On a Raspberry Pi 3 Model B v1.2 with
>   Linux raspberrypi 5.10.17-v7+ #1403 SMP Mon [...]

Hello. I have the same problem on Raspberry Pi 4B+ with Raspberry Pi OS (release date: 15.03.2024) when performing identical actions. I would be extremely grateful for a detailed solution. Have you found it?

If you don`t remember - https://coral.ai/docs/enviro-board/get-started/