f0cal/google-coral #96: cannot run mdt after successfully installing

tomnguyenn, 2021-02-16:
* Doc you were trying to follow: https://coral.ai/docs/dev-board-mini/get-started/#2-install-mdt
* Your host OS: macOS 11.2.1 Big Sur (M1 Macbook Air)
* Your Python3 version: 3.8.5

# What I ran

mdt

# What the docs said should happen

should show attached devices

# What actually happened

This error:

`Couldn't load paramiko or zeroconf -- perhaps you need to install them?
On Debian derivatives, 'apt-get install python3-paramiko python3-zeroconf'.`

I have confirmed with pip3 freeze that paramiko==2.7.2 and zeroconf==0.28.8 are indeed installed

I wonder if this is an issue with apple silicon or if other people have fun into the same issue...

tomnguyenn, 2021-02-26:
Just bumping this. Has anyone had any luck with an M1 Mac?

dtlam26, 2021-07-26:
A quick solve for this. If you look at the error raise in  ['main python file'](https://coral.googlesource.com/mdt/+/refs/heads/master/mdt/main.py). You will see the problem is that you cant import  **zeroconf** or **paramiko**.

Please import them manually in the python shell to see what library is conflicted or failed. For my case, it is PyNaCl missing. Therefore, just install/reinstall the library to fix the import

> Type "help", "copyright", "credits" or "license" for more information.
> >>> import paramiko
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/home/dtlam26/.local/lib/python3.7/site-packages/paramiko/__init__.py", line 22, in <module>
>     from paramiko.transport import SecurityOptions, Transport
>   File "/home/dtlam26/.local/lib/python3.7/site-packages/paramiko/transport.py", line 90, in <module>
>     from paramiko.ed25519key import Ed25519Key
>   File "/home/dtlam26/.local/lib/python3.7/site-packages/paramiko/ed25519key.py", line 22, in <module>
>     import nacl.signing
>   File "/usr/lib/python3/dist-packages/nacl/signing.py", line 19, in <module>
>     import nacl.bindings
>   File "/usr/lib/python3/dist-packages/nacl/bindings/__init__.py", line 17, in <module>
>     from nacl.bindings.crypto_box import (
>   File "/usr/lib/python3/dist-packages/nacl/bindings/crypto_box.py", line 18, in <module>
>     from nacl._sodium import ffi, lib
> ModuleNotFoundError: No module named 'nacl._sodium'