f0cal/google-coral #97: winpty: error: cannot start 'mdt': Not found in PATH

Tylersuard, 2021-02-24:
* Doc you were trying to follow: https://coral.ai/docs/dev-board-mini/get-started/#2-install-mdt
* Your host OS: Windows 10
* Your Python3 version: python 3.8

# What I ran
echo "alias mdt='winpty mdt'" >> ~/.bash_profile

source ~/.bash_profile

mdt devices

# What the docs said should happen

It should show devices

# What actually happened

winpty: error: cannot start 'mdt': Not found in PATH

noahzhy, 2021-02-26:
```shell
pip3 uninstall mendel-development-tool
pip3 install mendel-development-tool
```
Notice: pip3 install it without `--user`, btw, make sure the following content in `~/.bash_profile` file.

```shell
alias python3='winpty python3.exe'
alias mdt='winpty mdt'
```
Remember `source ~/.bash_profile`

loiswen, 2023-04-12:
Hello, I have the same problem, but it is not solved after trying your method, I am a newbie, here is my code process, is there something wrong with me?
`17272@DESKTOP-FEIFH6G MINGW64 ~
$ python3 -m pip install --user mendel-development-tool
Requirement already satisfied: mendel-development-tool in c:\users\17272\appdata\ro
aming\python\python38\site-packages (1.5.2)
Requirement already satisfied: zeroconf>=0.27.0 in c:\users\17272\appdata\roaming\p
ython\python38\site-packages (from mendel-development-tool) (0.47.3)
Requirement already satisfied: paramiko>=2.0.0 in c:\users\17272\appdata\roaming\py
thon\python38\site-packages (from mendel-development-tool) (3.1.0)
Requirement already satisfied: bcrypt>=3.2 in c:\users\17272\appdata\roaming\python
\python38\site-packages (from paramiko>=2.0.0->mendel-development-tool) (4.0.1)
Requirement already satisfied: cryptography>=3.3 in c:\users\17272\appdata\roaming\
python\python38\site-packages (from paramiko>=2.0.0->mendel-development-tool) (39.0
.2)
Requirement already satisfied: pynacl>=1.5 in c:\users\17272\appdata\roaming\python
\python38\site-packages (from paramiko>=2.0.0->mendel-development-tool) (1.5.0)
Requirement already satisfied: ifaddr>=0.1.7 in c:\users\17272\appdata\roaming\pyth
on\python38\site-packages (from zeroconf>=0.27.0->mendel-development-tool) (0.2.0)
Requirement already satisfied: async-timeout>=3.0.0 in c:\users\17272\appdata\roami
ng\python\python38\site-packages (from zeroconf>=0.27.0->mendel-development-tool) (
4.0.2)
Requirement already satisfied: cffi>=1.12 in c:\users\17272\appdata\roaming\python\
python38\site-packages (from cryptography>=3.3->paramiko>=2.0.0->mendel-development
-tool) (1.15.1)
Requirement already satisfied: pycparser in c:\users\17272\appdata\roaming\python\p
ython38\site-packages (from cffi>=1.12->cryptography>=3.3->paramiko>=2.0.0->mendel-
development-tool) (2.21)

17272@DESKTOP-FEIFH6G MINGW64 ~
$ echo "alias mdt='winpty mdt'" >> ~/.bash_profile

source ~/.bash_profile

17272@DESKTOP-FEIFH6G MINGW64 ~
$ mdt devices
winpty: error: cannot start 'mdt': Not found in PATH
17272@DESKTOP-FEIFH6G MINGW64 ~
$ pip3 uninstall mendel-development-tool
Found existing installation: mendel-development-tool 1.5.2
Uninstalling mendel-development-tool-1.5.2:
  Would remove:
    c:\users\17272\appdata\roaming\python\python38\scripts\mdt.exe
    c:\users\17272\appdata\roaming\python\python38\site-packages\mdt\*
    c:\users\17272\appdata\roaming\python\python38\site-packages\mendel_development_tool-1.5.2.dist-info\*
    c:\users\17272\appdata\roaming\python\share\man\man1\mdt.1
Proceed (Y/n)? Y
  Successfully uninstalled mendel-development-tool-1.5.2

17272@DESKTOP-FEIFH6G MINGW64 ~
$ pip3 install mendel-development-tool
Defaulting to user installation because normal site-packages is not writeable
Collecting mendel-development-tool
  Using cached mendel_development_tool-1.5.2-py3-none-any.whl (26 kB)
Requirement already satisfied: zeroconf>=0.27.0 in c:\users\17272\appdata\roaming\python\python38\site-packages (from mendel-development-tool) (0.47.3)
Requirement already satisfied: paramiko>=2.0.0 in c:\users\17272\appdata\roaming\python\python38\site-packages (from mendel-development-tool) (3.1.0)
Requirement already satisfied: pynacl>=1.5 in c:\users\17272\appdata\roaming\python\python38\site-packages (from paramiko>=2.0.0->mendel-development-tool) (1.5.0)
Requirement already satisfied: bcrypt>=3.2 in c:\users\17272\appdata\roaming\python\python38\site-packages (from paramiko>=2.0.0->mendel-development-tool) (4.0.1)
Requirement already satisfied: cryptography>=3.3 in c:\users\17272\appdata\roaming\python\python38\site-packages (from paramiko>=2.0.0->mendel-development-tool) (39.0.2)
Requirement already satisfied: async-timeout>=3.0.0 in c:\users\17272\appdata\roaming\python\python38\site-packages (from zeroconf>=0.27.0->mendel-development-tool) (4.0.2)
Requirement already satisfied: ifaddr>=0.1.7 in c:\users\17272\appdata\roaming\python\python38\site-packages (from zeroconf>=0.27.0->mendel-development-tool) (0.2.0)
Requirement already satisfied: cffi>=1.12 in c:\users\17272\appdata\roaming\python\python38\site-packages (from cryptography>=3.3->paramiko>=2.0.0->mendel-development-tool) (1.15.1)
Requirement already satisfied: pycparser in c:\users\17272\appdata\roaming\python\python38\site-packages (from cffi>=1.12->cryptography>=3.3->paramiko>=2.0.0->mendel-development-tool) (2.21)
Installing collected packages: mendel-development-tool
  WARNING: The script mdt.exe is installed in 'C:\Users\17272\AppData\Roaming\Python\Python38\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed mendel-development-tool-1.5.2

17272@DESKTOP-FEIFH6G MINGW64 ~
$ alias python3='winpty python3.exe'

17272@DESKTOP-FEIFH6G MINGW64 ~
$ alias mdt='winpty mdt'

17272@DESKTOP-FEIFH6G MINGW64 ~
$ source ~/.bash_profile

17272@DESKTOP-FEIFH6G MINGW64 ~
$ mdt devices
winpty: error: cannot start 'mdt': Not found in PATH

17272@DESKTOP-FEIFH6G MINGW64 ~
$ echo "alias mdt='winpty mdt'" >> ~/.bash_profile

source ~/.bash_profile

17272@DESKTOP-FEIFH6G MINGW64 ~
$ mdt devices
winpty: error: cannot start 'mdt': Not found in PATH

`

noahzhy, 2023-04-12:
maybe you should check your `~/.bash_profile` file content

brianthelion, 2023-04-12:
@loiswen Your use of `pip install --user` is causing the `mdt` executable to be created where `winpty` can't find it. The provided solution will not work with `--user`.

loiswen, 2023-04-13:
> maybe you should check your `~/.bash_profile` file content

Hello, thank you for your reply. I saw in directory “C:\Users\17272” that the content in file “.bash_profile” is alias python3='winpty python3.exe' alias python3='winpty python3.exe' alias mdt='winpty mdt' alias python3='winpty python3.exe' alias python3='winpty python3.exe' alias mdt='winpty mdt' alias mdt='winpty mdt' alias python3='winpty python3.exe' alias mdt='winpty mdt'
I think I followed the instructions to every step, but I don't know why I still haven't achieved the desired results.

loiswen, 2023-04-13:
> maybe you should check your `~/.bash_profile` file content

Hello, thank you for your reply. I saw in directory “C:\Users\17272” that the content in file “.bash_profile” is` alias python3='winpty python3.exe' alias python3='winpty python3.exe' alias mdt='winpty mdt' alias python3='winpty python3.exe' alias python3='winpty python3.exe' alias mdt='winpty mdt' alias mdt='winpty mdt' alias python3='winpty python3.exe' alias mdt='winpty mdt'`
I think I followed the instructions to every step, but I don't know why I still haven't achieved the desired results.