f0cal/google-coral #98: MongoDB can't be installed

slbejaranom, 2021-03-01:
Hello everyone, I have a server using nodejs mongo and dlib for tracking and counting cars, sending some info to mongo, however I have this running in docker, although I can get it out of docker it seems impossible to install mongo, i've tried some tricks from raspberry but still can't manage to do it, any help would be appreciated.

brianthelion, 2021-03-02:
Share some example code, please.

slbejaranom, 2021-03-02:
There's not much, I managed to install docker and docker-compose in the coral dev board and was building the app again since if I just do docker-compose up it'll run with the same characteristics as host:

This is my docker-compose.yml

version: "3"

services: 
  interfaz-web:
    build: ./webPage
    volumes:
      - ./webPage:/usr/src/app      
      - "/etc/timezone:/etc/timezone:ro"
      - "/etc/localtime:/etc/localtime:ro"

    ports:
      - 3000:3000
    depends_on: 
      - scriptfinal
      - mongo

  scriptfinal:
    build: ./scriptService
    volumes:
      - ./scriptService:/usr/src/scriptService      
      - "/etc/timezone:/etc/timezone:ro"
      - "/etc/localtime:/etc/localtime:ro"
    devices:
      - /dev/video0:/dev/video0
      - /dev/bus/usb:/dev/bus/usb
      - /tmp/.X11-unix:/tmp/.X11-unix      
    ports:
      - 3500:3500
    privileged: true
    depends_on: 
      - mongo

  mongo: 
    container_name: mongo
    image: mongo
    ports:
      - "6000:27017"
    logging: 
      driver: none

brianthelion, 2021-03-02:
Error logs?