f0cal/google-coral #102: Windows OS: Connected dev board not visible in mdt using Git Bash

trezbit, 2021-06-06:
* Doc you were trying to follow: https://coral.ai/docs/dev-board-mini/get-started/#4-connect-to-the-boards-shell-via-mdt
* Your host OS: Windows 10 (WSL2)
* Your Python3 version: 3.9

# What I ran
 Got For Windows - Git Bash 2.31.1
1. Installed mdt and confirmed it is available via Git Bash following suggestions in https://github.com/f0cal/google-coral/issues/97 to address path issues
2. Connected the dev board via USB, confirmed a new Network Adapter (Remote NDIS Compatible Device) shows up in Device Manager
3. Ran `mdt devices`

# What the docs said should happen
Connected dev board should be listed 

# What actually happened
After some wait nothing gets listed (mdt cannot see the connected device)
Running `mdt wait-for-device` stalls in "Waiting for device..."

GokulSudheesh, 2021-08-04:
I had the same issue and this worked for me.

First connect your board to your PC.

Go to **Control Panel > Network and Internet > Network and Sharing Center**

Here you will see an **Unidentified network** click on it and go to **Properties**.
In properties, go to **Sharing** and check **Allow other network users to connect through this computer's internet connection**.

Now try running ```mdt shell``` command.
After you have successfully connected your board, make sure to uncheck the previous option in **Network Sharing**.

nrg0004, 2021-12-01:
I did the above but it did not fix my problem. Here is what fixed it - 

Apparently the dev board is seen as a unidentified network under public networks. My Windows Firewall settings blocked python.exe from writing to or reading to public networks. I did the following -

**How to check if Windows Firewall is blocking a program**

- Press Windows Key + R to open Run.
- Type control and press OK to open Control Panel.
- Click on System and Security.
- Click on Windows Defender Firewall.
- From the left pane Allow an app or feature through Windows Defender Firewall.
- In the allowed apps window, scroll through all the apps.
- Locate the app you want to review and see if the app is checked. **_Here the python.exe was not checked. I changed the setting and voila, mdt devices finds the device and everything works on from here._**
- If it is unchecked, the app is blocked on the Firewall.
- If your program is blocked, simply check the app and click OK.

jogugil, 2025-03-30:
Hello everyone,

I'm new to the Raspberry Pi Coral Dev Board and I'm encountering an issue where the board is recognized as NDIS but I'm unable to connect using mdt devices in WSL2. I've already enabled NDIS sharing as suggested, but I'm still unable to get it to work.

I've tried the steps mentioned earlier about network sharing, but no luck so far. Any additional suggestions or things I might be missing?

Thanks in advance for your help!