f0cal/google-coral #104: Coral Dev Board Mini not recognized in my Ubuntu laptop

PhanTask, 2021-11-12:
* Doc you were trying to follow: https://coral.ai/docs/dev-board-mini/get-started/#4-connect-to-the-boards-shell-via-mdt
* Your host OS: Ubuntu 20.04.3 LTS
* Your Python3 version: Python 3.8

# What I ran

`mdt devices`

# What the docs said should happen

`fun-zebra        (192.168.100.2)`

# What actually happened

It returned nothing.

I also tried running `mdt shell` and it said "Waiting for a device... Unable to find any devices on your local network segment."

It looks like the mini device was not recognized in my Ubuntu laptop.

I connected the device and my laptop using USB cable, and I also powered the device using another USB adaptor. I am sure the connection is correct (power port on the left side, and data port on the right, as described in the tutorial). The light on mini device is green.

Did I miss something?

nrg0004, 2021-11-18:
I also have the same problem but on my windows laptop. I even tried doing this - https://github.com/f0cal/google-coral/issues/102#issuecomment-892850619 but nothing helps. Did you find a solution?

PhanTask, 2021-11-29:
@nrg0004 Unfortunately, no... I don't know if it is caused by wrong configuration or invisible physical damage... No idea how to check.

nrg0004, 2021-11-30:
I have been able to connect to the dev board using serial console over micro-USB but it fails to connect with MDT over USB-C. I connected several boards through serial console but not through MDT over USB-C. The first board we connected with my colleagues computer successfully but it wouldn't connect with mine. Later figured that you can't connect one board to multiple computers unless you copy the key from the first computer to the others. We did that but still the problem remains with old and new boards. It cannot identify any devices. I am beginning to believe that the problem is with the computer not recognizing the network device over the USB-C.

nrg0004, 2021-12-01:
Finally I found a fix. Apparently the dev board is seen as a unidentified network under public networks. My Windows Firewall settings blocked python.exe from writing to or reading to public networks. I did the following -
**How to check if Windows Firewall is blocking a program**

- Press Windows Key + R to open Run.
- Type control and press OK to open Control Panel.
- Click on System and Security.
- Click on Windows Defender Firewall. 
- From the left pane Allow an app or feature through Windows Defender Firewall.
- In the allowed apps window, scroll through all the apps.
- Locate the app you want to review and see if the app is checked. **Here the python.exe was not checked. I changed the setting and voila, mdt devices finds the device and everything works on from here.** 
- If it is unchecked, the app is blocked on the Firewall.
- If your program is blocked, simply check the app and click OK.

PhanTask, 2021-12-14:
@nrg0004 Thanks sooooo much! Your suggestion definitely works for me. I tried connecting the dev board mini and windows desktop and enabled python.exe access, then I can use mdt shell to connect to the dev board. Awesome!

Just another question regarding private/public key thing. I have another coral dev board mini and it is very likely to be set up with another computer before. I tried connecting it using my own computer (windows, linux, or MacBook) and I cannot have access to it. According to your analysis, we need to copy the key from the old computer to the new computer. However, I do not have access to the old computer (I don't even know which one it is). Do you by chance have any solutions for this (e.g., recover/flash the board without having access to it)?