f0cal/google-coral #108: 7: Run our demo app - problem

sachlj, 2022-07-10:
sachlj@google-coral:~$ edgetpu_demo --device
Press 'q' to quit.
Press 'n' to switch between models.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 791, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (tflite-runtime 2.1.0.post1 (/usr/local/lib/python3.7/dist-packages), Requirement.parse('tflite-runtime==2.5.0.post1'), {'pycoral'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/edgetpu_detect", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3191, in <module>
    @_call_aside
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3175, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3204, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 585, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 598, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'tflite-runtime==2.5.0.post1' distribution was not found and is required by pycoral

sachlj@google-coral:~$ cat /proc/version
Linux version 4.14.98-imx (pbuilder@linux-imx-7c598330-766d-4b03-a12e-333509ed9db6-d1dc9-rtjkh) (gcc version 8.3.0 (Debian 8.3.0-2)) #1 SMP PREEMPT Tue Nov 2 02:55:21 UTC 2021
sachlj@google-coral:~$ gcc --version
gcc (Debian 8.3.0-6) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

sachlj@google-coral:~$ /lib/aarch64-linux-gnu/libc.so.6
GNU C Library (Debian GLIBC 2.28-10) stable release version 2.28.
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 8.3.0.
libc ABIs: UNIQUE ABSOLUTE
For bug reporting instructions, please see:
<http://www.debian.org/Bugs/>.
sachlj@google-coral:~$ cat /etc/debian_version
10.0
sachlj@google-coral:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Mendel
Description:    Mendel GNU/Linux 5 (Eagle)
Release:        10.0
Codename:       eagle

brianthelion, 2022-07-10:
It looks like the original error was with the `requirements.txt`

```
pkg_resources.ContextualVersionConflict: (tflite-runtime 2.1.0.post1 (/usr/local/lib/python3.7/dist-packages), Requirement.parse('tflite-runtime==2.5.0.post1'), {'pycoral'})
```

Can you please post a reference to the `edgetpu_demo` source code? Thanks!

sachlj, 2022-07-10:
**/usr/bin/edgetpu_demo

#!/bin/bash

readonly TEST_DATA="/usr/share/edgetpudemo"
readonly VIDEO_DEVICE_FILE="${TEST_DATA}/video_device.mp4"
readonly VIDEO_STREAM_FILE="${TEST_DATA}/video_stream.mp4"
readonly TPU_MODEL_FILE="${TEST_DATA}/mobilenet_ssd_v1_coco_quant_postprocess_edgetpu.tflite@Running MobileNet SSD v1 on Edge TPU"
readonly CPU_MODEL_FILE="${TEST_DATA}/mobilenet_ssd_v1_coco_quant_postprocess.tflite@Running MobileNet SSD v1 on CPU"
readonly LABELS_FILE="${TEST_DATA}/coco_labels.txt"

if [[ "$1" == "--device" ]]; then
  echo "Press 'q' to quit."
  echo "Press 'n' to switch between models."

  edgetpu_detect \
      --source "${VIDEO_DEVICE_FILE}" \
      --model "${TPU_MODEL_FILE},${CPU_MODEL_FILE}" \
      --labels "${LABELS_FILE}" \
      --filter car,truck \
      --max_area 0.1 \
      --color white \
      --loop \
      --displaymode fullscreen
elif [[ "$1" == "--stream" ]]; then
  echo "Press 'q' to quit."
  echo "Press 'n' to switch between models."

  SERVER_INDEX_HTML="${TEST_DATA}/index.html" edgetpu_detect_server \
      --source "${VIDEO_STREAM_FILE}" \
      --model "${TPU_MODEL_FILE},${CPU_MODEL_FILE}" \
      --labels "${LABELS_FILE}" \
      --filter car,truck \
      --max_area 0.1 \
      --color white \
      --loop
else
  echo "Run on-device inference:"
  echo "  $0 --device"
  echo "Run streaming server:"
  echo "  $0 --stream"
fi

sachlj, 2022-07-10:
https://coral.ai/docs/dev-board/get-started#run-demo
.......
sachlj@google-coral:~$ edgetpu_demo --stream
Press 'q' to quit.
Press 'n' to switch between models.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 791, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (tflite-runtime 2.1.0.post1 (/usr/local/lib/python3.7/dist-packages), Requirement.parse('tflite-runtime==2.5.0.post1'), {'pycoral'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/edgetpu_detect_server", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3191, in <module>
    @_call_aside
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3175, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3204, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 585, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 598, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'tflite-runtime==2.5.0.post1' distribution was not found and is required by pycoral
sachlj@google-coral:~$

sachlj, 2022-07-10:
8: Run a model using the PyCoral API
....
sachlj@google-coral:~$ cd coral
sachlj@google-coral:~/coral$ cd pycoral
sachlj@google-coral:~/coral/pycoral$ bash examples/install_requirements.sh classify_image.py
DOWNLOAD: mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 4197k  100 4197k    0     0  12.1M      0 --:--:-- --:--:-- --:--:-- 12.1M
DOWNLOAD: mobilenet_v2_1.0_224_inat_bird_quant.tflite
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 3495k  100 3495k    0     0  10.4M      0 --:--:-- --:--:-- --:--:-- 10.4M
DOWNLOAD: inat_bird_labels.txt
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 37146  100 37146    0     0   152k      0 --:--:-- --:--:-- --:--:--  152k
DOWNLOAD: parrot.jpg
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  582k  100  582k    0     0  2268k      0 --:--:-- --:--:-- --:--:-- 2268k
sachlj@google-coral:~/coral/pycoral$ python3 examples/classify_image.py \
> --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
> --labels test_data/inat_bird_labels.txt \
> --input test_data/parrot.jpg
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/tflite_runtime/interpreter.py", line 161, in load_delegate
    delegate = Delegate(library, options)
  File "/usr/local/lib/python3.7/dist-packages/tflite_runtime/interpreter.py", line 120, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/classify_image.py", line 121, in <module>
    main()
  File "examples/classify_image.py", line 71, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "/usr/local/lib/python3.7/dist-packages/tflite_runtime/interpreter.py", line 164, in load_delegate
    library, str(e)))
ValueError: Failed to load delegate from libedgetpu.so.1

sachlj@google-coral:~/coral/pycoral$

sachlj, 2022-07-10:
Shouldn't Google's tutorial for their product work by any chance? Or does Google produce garbage too?

brianthelion, 2022-07-10:
> Shouldn't Google's tutorial for their product work by any chance?

The reason this community issue tracker exists is that Google is doing a questionable job maintaining the public-facing Coral software.

jlpoolen, 2022-07-10:
> Shouldn't Google's tutorial for their product work by any chance? Or does Google produce garbage too?

_Ad hominem_ attacks, "Google produce garbage", are an unfortunate way to achieve the goal of getting their demo software working.  If you are trying to get their Dev Board to work with the demo software outside of its configured demonstration, good luck.  I've spent a considerable amount of time studying their code and effort to get gstreamer working (to the point of installing a Gentoo chroot environment with a high watermark of gstreamer, and trying to make it work, and have concluded that for the Dev Board, anything outside of the sample is going to be risky business.  I was able to get some MP4 files to load, e.g., such as those Reolink generates, but they had to be reprocessed in order to not be rejected by the demo implementation of gstreamer.  I determined that the video samples themselves, while taken from the same source, a German photographer's duly and appropriate license that allows Google to redistribute his work, were different depending on which options you feed to the software.  This subtle distinction is nowhere disclosed.  

I decided to no long try to make the Dev Board do anything more than demonstrate a proof of concept and I am awaiting the arrival of the USB version which has been back-ordered at UK Farnell for months and was estimated to be shipping in June -- which it hasn't.  I ran into memory problems using my Gentoo chroot version despite giving lots of swap space.  

The Coral chip is a marvelous innovation, it simply is not available at this time, July 2022, and I am waiting with great expectations the arrival of the USB version so I can implement it on a Linux system I have complete control of, e.g. Gentoo.

Oh, and Mendel innovation has ceased, so working within the Mendel environment is now relegated to the category of historic interest.  I would say the Google Team became very much aware of the limitations of the Dev Board and therefor introduced the USB version so its true virtues could be harnessed in a suitable environment by patient and determined software developers.